tu-darmstadt-ros-pkg/hector_slam #16: Bug in interpMapValueWithDerivatives (OccGridMapUtil.h)?

alik55, 2015-03-26:
It seems x and y weights are switched in derivatives at Line 341:

 return Eigen::Vector3f(
      ((intensities[0] * xFacInv + intensities[1] * factors[0]) * (yFacInv)) +
      ((intensities[2] * xFacInv + intensities[3] * factors[0]) * (factors[1])),
      -((dx1 * xFacInv) + (dx2 * factors[0])),
      -((dy1 * yFacInv) + (dy2 * factors[1]))

should be (changes in the last two lines):

 return Eigen::Vector3f(
      ((intensities[0] * xFacInv + intensities[1] * factors[0]) * (yFacInv)) +
      ((intensities[2] * xFacInv + intensities[3] * factors[0]) * (factors[1])),
      -((dx1 * yFacInv) + (dx2 * factors[1])),
      -((dy1 * xFacInv) + (dy2 * factors[0]))

BossKwei, 2016-07-31:
Hi,
I also find that bug these days, and I'm confused why there is a math error but this lib could run correctly.
I think that because _xFacInv_ or _yFacInv_ are both weight value in range (0,1), and dx,dy decide the positive or negative. A multiplication of _xFacInv_*dx1 is same to k * dx1 * _yFacInv_.

watcherZY, 2018-05-15:
i have confirmed that you are correct , i used a same rosbag to test  -((dx1 * xFacInv) + (dx2 * factors[0])),
-((dy1 * yFacInv) + (dy2 * factors[1]))  AND -((dx1 * yFacInv) + (dx2 * factors[1])),
-((dy1 * xFacInv) + (dy2 * factors[0])).   the second one works much better! and dosen^t  become disorder when the robot rotation

jondo, 2021-03-17:
Why was this closed?

StefanFabian, 2021-03-17:
I've closed all issues older than 5 years due to their age but you're right, this one may still be relevant.
I'm not sure if @skohlbr will have the time to look at this, though.

HX2003, 2025-04-04:
This seems to resolve the segmentation fault issue. I think it happens because somehow the coordinates provided to interpMapValueWithDerivatives become NaN, concreteGridMap->pointOutOfMapBounds(coords) does not detect NaN value.