tu-darmstadt-ros-pkg/hector_slam #26: Moving map around robot pose

aswinhope, 2016-01-29:
Is it possible to move the start position of the map in realtime such that the map generated with always have its center as the robot pose? -- something like the costmap in move base. At the moment if the robot goes beyond the limits of the map size, it loses localization.

skohlbr, 2016-02-02:
No, this is currently not possible. With some effort some sort of moving map functionality (either growing map tiles behind the scenes or copying around maps in double buffer fashion) would be possible, but there currently are no plans to implement this from my side.

aswinhope, 2016-02-09:
Thank you Stefan. At the moment, I'm using a huge map and periodically clearing out the old cells far from the robot. This does have a drawback of using up a lot of RAM. A moving map may not be so practical to do in realtime -- probably the concept of submaps is better

meyerj, 2016-02-09:
You could give [hector_mapping2](https://github.com/tu-darmstadt-ros-pkg/hector_mapping2) a try. The package implements the same algorithm as hector_mapping in this repo, but it also supports tiled 2D and 3D maps. It is not very well tested, so do not expect good and stable results with the current development version.

As with hector_mapping, the map origin is the initial pose of the robot when mapping was started, but the map grows as soon as a laser scan reaches the current boundaries. Not sure if the current costmap_2d package in the navigation stack can handle this - at the time I was working on hector_mapping2 I needed a patched version to allow for map size changes.

aswinhope, 2016-02-09:
Cool will check it out. Thanks

hashim19, 2016-04-19:
how can i use this package ?

closed 2021-03-17 by StefanFabian