tu-darmstadt-ros-pkg/hector_slam #30: results are slightly different between deb and compiled node

k-okada, 2016-07-27:
hi,

I have noticed that result of deb node and compiled node is slightly different (c.f. https://github.com/tu-darmstadt-ros-pkg/hector_slam/issues/6), it seems deb one is better. does any one   know why?

deb node:
[image]

catkin compiled node:
[image]

meyerj, 2016-07-27:
You already cited issue #6, which would be my first guess of what could be different. The ROS debians are built in release mode with the `-DNDEBUG` compile option, similar to cmake's `Release` or `RelWithDebInfo` build types.

Did you compile your catkin workspace with `-DCMAKE_BUILD_TYPE=Release`?

However, we could add a line to `hector_mapping/CMakeLists.txt` that sets the build type to `Release` for hector_mapping only unless set explicitly by the user.

k-okada, 2016-07-29:
ok, I created PR to compile with optimization flag, this will generate same result for both deb and source -> https://github.com/tu-darmstadt-ros-pkg/hector_slam/pull/31

closed 2016-07-30 by k-okada