tu-darmstadt-ros-pkg/hector_slam #33: problem with map updates

glebos, 2016-09-02:
Hi,
I am trying to simulate a car-like robot in gazebo and to create a map of the world by using hector mapping. When I try to run hector mapping, the map is not updating, like this: https://postimg.org/image/c34oos2u9/ https://postimg.org/image/9n2vaxkrl/.
I don't get any errors and cannot figure out what is the problem.
This is my tf tree: https://postimg.org/image/yzrily0o3/.
Running roswtf ouputs:
```
WARNING The following node subscriptions are unconnected:
 * /hector_mapping:
   * /syscommand
 * /rviz_1472805256594766659:
   * /map_updates
 * /gazebo:
   * /gazebo/set_model_state
   * /gazebo/set_link_state

WARNING These nodes have died:
 * my_chassis/urdf_spawnermy_chassis-3
```
Maybe the problem is that map_updates_node of rviz is not connected?
Any suggestions how to solve the problem will be greatly appreciated.

Thanks in advance!

skohlbr, 2016-09-02:
Does the robot move in rviz when you drive it around in Gazebo? What are your map and scan topics? What is the console output (if any) of hector_mapping?

glebos, 2016-09-02:
Yes, the robot is moving in rviz, but the map is not updating. My map topic is /map and my scan topic is /my_chassis/scan. When running the mapping launch file I get following ouput: lookupTransform my_chassis/base_footprint to my_chassis/hokuyo_link timed out. Could not transform laser scan into base_frame. It just appears in the beginning 2 times. Then it stops to publish this message.

skohlbr, 2016-09-02:
That sounds ok then. Can you make a bag-file? Scan and tf data should be sufficient.

glebos, 2016-09-06:
Do you want me to upload the bagfile?

skohlbr, 2016-09-06:
Yes :)

glebos, 2016-09-06:
I have uploaded the file here: https://ufile.io/5cb30. By the way, gmapping is running correctly with that.

skohlbr, 2016-09-06:
Map is updating for me, but the LIDAR has max range scan points showing up as valid ranges, which messes up scan matching. There often are very few features in the scans however, so it is unlikely that such a data set can produce good results.

glebos, 2016-09-06:
could you please post your launch file? since I cannot get hector mapping to run correctly. Do you think the problem is with the LIDAR? I am using the hokuyo gazebo plugin...

skohlbr, 2016-09-06:
The range is set very low and the implementation of the gazebo hokuyo plugin appears to be buggy, making max range scan point appear as valid end points (so the robot always sees a circle around itself). I removed that effect by reducing the allowed laser max range. See changeset here: https://github.com/tu-darmstadt-ros-pkg/hector_slam/commit/9faf097ed1fd98be294064cc1c3825bb2089f92b

If you increase the range of the simulated LIDAR it might work better. I also noticed weird artifacts popping up in scans every once in a while, not sure where those come from.

glebos, 2016-09-06:
I have set the range of the laser to 6m, because I am trying to simulate a rplidar. Why do you assume that the implementation of the hokuyo plugin is buggy?

skohlbr, 2016-09-06:
Because a real LIDAR does not generate a circle of beam end points around the robot when there are no obstacles within max range.

glebos, 2016-09-06:
It's strange that I can't create the map using the same launch and bag file. I don't get any result

skohlbr, 2016-09-07:
Did you set the parameter `use_sim_time` to `true` and played back the bag with the `--clock` option? If not, you should do so :)

glebos, 2016-09-08:
Oh yeah, the sim_time was the problem apparently. Thank you very much.
I am now able to create a map, but for some reason not the full range of the laser scan is used to create the map: https://postimg.org/image/onbpc4j5x/. Do you have an idea how to fix that?

closed 2016-09-27 by glebos