tu-darmstadt-ros-pkg/hector_slam #35: Huge covariance values in /poseupdate

ronwalf, 2016-10-25:
I'm getting huge covariance values from hector_mapping, both on my platform and when playing bag files with the tutorial launch files.

For instance with
```
$ roslaunch hector_slam_launch tutorial.launch &
$ rosbag play Team_Hector_MappingBox_RoboCup_2011_Rescue_Arena.bag --clock &
$ rostopic echo /poseupdate 
```
I get:
```
---
header: 
  seq: 1092
  stamp: 
    secs: 1310298019
    nsecs: 658137722
  frame_id: map
pose: 
  pose: 
    position: 
      x: 23.9624099731
      y: 17.8734512329
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.299876102962
      w: 0.953978156392
  covariance: [
    96.53327941894531, -47.63311767578125, 0.0, 0.0, 0.0, -1181.0054931640625,
   -47.63311767578125, 97.9208984375, 0.0, 0.0, 0.0, 1569.2373046875,
   0.0, 0.0, 0.0, 0.0, 0.0, 0.0,
   0.0, 0.0, 0.0, 0.0, 0.0, 0.0,
   0.0, 0.0, 0.0, 0.0, 0.0, 0.0,
  -1181.0054931640625, 1569.2373046875, 0.0, 0.0, 0.0, 163802.53125]
---
```

meyerj, 2016-11-01:
We are violating the specification of the [geometry_msgs/PoseWithCovariance](http://docs.ros.org/api/geometry_msgs/html/msg/PoseWithCovariance.html) message here. What is output in the `poseupdate` message is not a covariance, but the Hessian calculated during the scan matching process (see [ScanMatcher.h:184](https://github.com/tu-darmstadt-ros-pkg/hector_slam/blob/571790634671b544d8dc2cdc97527213ecd70b2d/hector_mapping/include/hector_slam_lib/matcher/ScanMatcher.h#L184)). It can be interpreted as the inverse of the covariance matrix up to a constant factor that depends on the variance of the distance measurement of a single laser ray and the map resolution.

The Hessian (or information matrix) has the advantage that its diagonal elements go to zero for maximum uncertainty instead of having to represent this situation with -1 like in other occurrences of variances in ROS.

12tbuchman, 2017-03-17:
I am trying to use Hector SLAM as a visual odometry source with an IMU, both of which will pass through an EKF to provide the odometry for AMCL localization. However because we don't have a true covariance from Hector SLAM the sensor fusion is much more difficult. Is it possible to output the real covariance instead of the Hessian?

meyerj, 2017-04-25:
It would not be hard to add an extra parameter to select whether to output the uncertainty as a covariance (with the typical ROS convention to encode infinity by -1 on the diagonal elements) or information matrix (both have an unknown scale factor that depends on the accuracy of the input data).

I do not have time to implement this myself, but pull requests are definitely welcome (@skohlbr?).

doisyg, 2017-08-07:
@meyerj , so you are saying that if we inverse the 6x6 poseupdate covariance message, we get the covariance up to a constant factor? As simple as that?

meyerj, 2017-08-07:
Yes. But some components can have zero columns and rows, so you have to be careful to handle this case correctly. For example, you could invert the sliced matrix that only contains the rows and columns with non-zero diagonal elements and replace the others by -1 or Nan in the inverted matrix. Should be doable with some Eigen magic. Or simply set the whole covariance matrix to -1 if the Hessian is not invertible. It really depends on how the receiving end would interpret this. I am not aware of a ROS standard on how partial covariance matrices (or partially infinite variances) should be represented in messages.

That's exactly the reason why I think the information matrix is the more natural representation of uncertainty here and it can be directly used as a weighting matrix for fusion algorithms.

doisyg, 2017-08-08:
Thanks for your answer @meyerj .
I intended to use directly the output of the /poseupdate message as an input pose for the robot_localization package. But after thinking about it, as in practice some kind of covariance tuning is always needed for Kalman filtering, your idea of using it as a weighting matrix makes more sense.

meyerj, 2017-08-08:
Simply considering the map pose as a standard Kalman filtering measurement update is not recommended, because the pose errors in each iteration are not independent and hence violating the Kalman assumptions. [hector_localization](http://wiki.ros.org/hector_localization) therefore uses a different approach for pose updates based on [covariance intersection](https://en.wikipedia.org/wiki/Covariance_intersection), but there are other options. The same is actually true for a pose provided by a GNSS receiver, which is also a filtered result and errors are not independent over time. Not sure about the implementation of robot_localization in that respect.

closed 2017-08-08 by meyerj

doisyg, 2017-08-08:
I am aware that the map pose is violating the Kalman assumptions. However, using the yaw information in a differential way is showing very interesting result.