tu-darmstadt-ros-pkg/hector_slam #39: Does this work with ROS Kinetic and Ubuntu 16.04?

hjjayakrishnan, 2017-04-08:
If yes then which branch should be cloned and built? If there is some documentation on the installation, kindly share. Thanks

Userskii, 2017-05-09:
This may help.
http://answers.ros.org/question/244776/is-it-possible-to-run-the-hector_quadrotor-demos-in-kinetic/

Hossein76, 2017-07-13:
yes it does work with kinetic and Ubuntu 16.04
in order to install it you can use apt-get : 
`sudo apt-get install ros-kinetic-hector-slam`
next you can bring up different nodes of hector slam using either launch files or rosrun command in terminal
for hector mapping you can in a launch file insert this : 
`<node pkg="hector_mapping" type="hector_mapping" name="arbitrary_name_for_node" />`
or in the terminal after starting roscore type : 
`rosrun hector_mapping hector_mapping`
these two commands start hector mapping with all the parameters set to default values
for other packages in the hector slam you can use these to command but you just need to change the name of the package and the node you want to use.
if you want to install a single package you can use apt-get :
`sudo apt-get install ros-kinetic-hector-packagename`
and insert the name of desired package instead of packagename

watcherZY, 2018-05-15:
It works very well

closed 2019-05-18 by StefanFabian