tu-darmstadt-ros-pkg/hector_slam #40: Is the Hector mapping applicable for dynamic environment too?

engrmz, 2017-05-17:
Hi,
I am trying to use Hector mapping in dynamic environment. However, both of them are showing/mapping single obstacle twice in a map (i.e. before movement and after movement). Is there any way to get latest position (after movement) of the obstacle only? 
Thanks in advance.

engrmz, 2017-05-17:
[image]

[image]

Please see the attached images. Single obstacle is mapped twice by the robot while using hector mapping.

skohlbr, 2019-10-31:
There is not explicit consideration/modeling of dynamic obstacles, but the the probabilistic map cell update will update cells according to if free or occupied space has been seen by the sensor. Default settings are not very reactive, but can be changed via the `update_factor_free` and `update_factor_occupied` parameters described here: http://wiki.ros.org/hector_mapping#Parameters

closed 2019-10-31 by skohlbr

jjbecomespheh, 2020-07-07:
@engrmz Hi, may I check if you managed to implement it with Hector SLAM?

engrmz, 2020-07-07:
> @engrmz Hi, may I check if you managed to implement it with Hector SLAM?

Hello,
I used octo-mapping for static map creation. Then I used scan matching technique for detecting dynamic object.
You can see the results in the videos here!
[https://www.youtube.com/watch?v=CByqTLxnq2o&list=PL1bGrABXHXbgfiJIb2QirOkkajmpSxXkz&index=3](url)