tu-darmstadt-ros-pkg/hector_slam #47: Using Mapping with VLP16-Hires

KevinRiehl, 2018-02-14:
Dear Hector Team,
I have a question concerning your mapping package.

First of all I am using Ros Kinetic on Ubuntu 16.04.03 and
a Velodyne Lidar (VLP16 Hires).
Before I run your package I start the velodyne driver (https://github.com/ros-drivers/velodyne), so that
there is a pointcloud and a laserscan from the Lidar. It is visible
via RViz.

Then, running your package, I receive following error message:
`lookupTransform base_footprint to velodyne timed out. Could not transform laser scan intobase_frame.`

I looked on your code and this message is thrown because there is time out by
trying to use the tf "velodyne". It does not exist if you check it with "rosrun tf tf_monitor"
because the Velodyne Driver does not publish any tf called "velodyne".

What shell I do in order to get your code running?
Thanks,
Kevin Riehl

aybakana, 2018-03-28:
When I first used VLP16 on ROS, I used the command below but I couldnt see the point cloud.
`rosrun rviz rviz `
So I saw that in the [tutorials](http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20HDL-32E), they use the command below and then velodyne frame comes out.
`rosrun rviz rviz -f velodyne`
If you are using roslaunch, you can try adding "-f velodyne" as arguments like below.
<node pkg="rviz" type="rviz" name="rviz" args="-f velodyne"/>

strecksen, 2018-09-24:
Hi Kevin, I had the same problem.
Based on https://answers.ros.org/question/53611/hector_mapping-base_link-laser-transform-times-out-once/ and https://answers.ros.org/question/234365/hector_slam-not-working-with-my-bag-file/
I suggest to add in your launch file
`<node pkg="tf" type="static_transform_publisher" name="base_footprint_2_velodyne" args="0 0 0 0 0 0 /base_footprint /velodyne 10"/>`
If it fails, try to replace base_footprint through base_link or base_frame.

karry3775, 2019-01-04:
Hey Kevin I am also currently working on VLP 16 Hi Res, and using ROS Kinetic on Ubuntu  16.04.03.
I am following the ROS wiki tutorials to visualize the data.
Link: http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20Velodyne%20VLP16
However I am not able to get any data when I hit
rostopic echo /velodyne_points
And thus also not able to visualize any data on rviz.
It will be really great if you can help me with this.
Thank you!