tu-darmstadt-ros-pkg/hector_slam #48: Over a network /scan working but hector_mapping

SabyasachiDash, 2018-02-16:
I have correctly networked two ROS Kinetic laptops on LAN. One laptop,the MASTER, is for viewing and processing the info; the other laptop, the ROBOT CONTROLLER(RC), is on the robot. I'm using a RPLIDAR A2 on my robot. Its connected to the RC(Robot Controller), and is publishing over /scan topic. I'm able to view the data from /scan topic on my MASTER laptop, using Rviz. I'm then running the hector_mapping package on my MASTER, with the following mapping_default.launch file---
---------------------------------------------------------------
<?xml version="1.0"?>

<launch>
  <arg name="tf_map_scanmatch_transform_frame_name" default="scanmatcher_frame"/>
  <arg name="base_frame" default="base_frame"/>
  <arg name="odom_frame" default="base_frame"/>
  <arg name="pub_map_odom_transform" default="true"/>
  <arg name="scan_subscriber_queue_size" default="5"/>
  <arg name="scan_topic" default="scan"/>
  <arg name="map_size" default="2048"/>
  
  <node pkg="hector_mapping" type="hector_mapping" name="hector_mapping" output="screen">
    
    
    <param name="map_frame" value="map" />
    <param name="base_frame" value="$(arg base_frame)" />
    <param name="odom_frame" value="$(arg odom_frame)" />
    
    
    <param name="use_tf_scan_transformation" value="true"/>
    <param name="use_tf_pose_start_estimate" value="false"/>
    <param name="pub_map_odom_transform" value="$(arg pub_map_odom_transform)"/>
    
    
    <param name="map_resolution" value="0.050"/>
    <param name="map_size" value="$(arg map_size)"/>
    <param name="map_start_x" value="0.5"/>
    <param name="map_start_y" value="0.5" />
    <param name="map_multi_res_levels" value="2" />
    
    
    <param name="update_factor_free" value="0.4"/>
    <param name="update_factor_occupied" value="0.9" />    
    <param name="map_update_distance_thresh" value="0.4"/>
    <param name="map_update_angle_thresh" value="0.06" />
    <param name="laser_z_min_value" value = "-1.0" />
    <param name="laser_z_max_value" value = "1.0" />
    
     
    <param name="advertise_map_service" value="true"/>
    
    <param name="scan_subscriber_queue_size" value="$(arg scan_subscriber_queue_size)"/>
    <param name="scan_topic" value="$(arg scan_topic)"/>
    
    
    
   
  </node>
    
  <node pkg="tf" type="static_transform_publisher" name="map_nav_broadcaster" args="0 0 0 0 0 0 base_frame laser 100"/>
</launch>
  
  ---------------------------------------------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------------------------------------------------
I'm getting the following error-----------
lookupTransform base_frame to laser timed out. Could not transform laser scan into base_frame
-------------------------------------------------------
BUT THIS IS NOT IT!
When I'm running the same package on my RC(Robot Controller) Laptop with the RPLIDAR, I'm able to successfully publish over the LAN and view the map from my MASTER using Rviz.
Why isn't it working the other way around?
I've checked but I'm receiving the /scan data perfectly on my MASTER. I'm able to successfully run rplidar & hector_mapping together simultaneously on either MASTER or RC, but NOT when rplidar is on one laptop and hector_mapping on another.
I've searched the internet, wasn't able to find a solution.
Thanks in advance...

sjamthe, 2019-07-11:
If you look at /scan topic data the message produced has frame_id: "laser"
You should try to change the following lines
arg name="base_frame" default="base_frame"/>
<arg name="odom_frame" default="base_frame"/>

The values for base_frame and odom_frame should be 'laser'

This worked for me.

ACSLli, 2019-09-03:
Hi did you fix the problem? recently I also got that error. I guess it is because the rplidar scan points it is not enough. When I changed rplidar scan_mode to 4k, I got the problem. But when the scan_mode is 10k or 16k I did get that error.

IrekRybark, 2021-04-19:
In my case the issue was a time difference between two machines.  
After installing ntp and starting it, the message went away.

OGChadesh, 2021-04-30:
I experienced same issue while connecting to wifi network produced by raspberry pi3b. The time on the Pi differed from the time set on the laptop. Adjusting the time and timezone using timedatectl on the Pi resolved the issue. Thanks @IrekRybark .

kabilan2003, 2023-07-20:
> If you look at /scan topic data the message produced has frame_id: "laser" You should try to change the following lines arg name="base_frame" default="base_frame"/>
> 
> The values for base_frame and odom_frame should be 'laser'
> 
> This worked for me.

thank lot