tu-darmstadt-ros-pkg/hector_slam #54: Compatibility on ROS Melodic

BBB-noob, 2018-07-12:
Hi!

I am currently operating on Ubuntu 18.04 with ROS Melodic and I wanted to use the hector_slam package (or group of package). Since there is no release for Melodic, I tried to clone the files directly in my catkin package and make a catkin_make.

But I encounter errors when compiling:
beagle@beaglebone:~/catkin_project$ catkin_make --pkg hector_mapping
Base path: /home/beagle/catkin_project
Source space: /home/beagle/catkin_project/src
Build space: /home/beagle/catkin_project/build
Devel space: /home/beagle/catkin_project/devel
Install space: /home/beagle/catkin_project/install
Creating symlink "/home/beagle/catkin_project/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/beagle/catkin_project/src -DCATKIN_DEVEL_PREFIX=/home/beagle/catkin_project/devel -DCMAKE_INSTALL_PREFIX=/home/beagle/catkin_project/install -G Unix Makefiles" in "/home/beagle/catkin_project/build"
####
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/beagle/catkin_project/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/beagle/catkin_project/build/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.15") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 13 packages in topological order:
-- ~~  - hector_slam (metapackage)
-- ~~  - hector_slam_launch
-- ~~  - hector_map_tools
-- ~~  - hector_nav_msgs
-- ~~  - hector_geotiff
-- ~~  - hector_geotiff_plugins
-- ~~  - hector_marker_drawing
-- ~~  - hector_compressed_map_transport
-- ~~  - hector_imu_attitude_to_tf
-- ~~  - hector_imu_tools
-- ~~  - hector_map_server
-- ~~  - hector_trajectory_server
-- ~~  - hector_mapping
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 80 lines cut ...]
Scanning dependencies of target hector_mapping_generate_messages_lisp
[ 56%] Generating Lisp code from hector_mapping/HectorIterData.msg
[ 62%] Generating Lisp code from hector_mapping/HectorDebugInfo.msg
[ 62%] Built target hector_mapping_generate_messages_lisp
Scanning dependencies of target hector_mapping_generate_messages_nodejs
[ 68%] Generating Javascript code from hector_mapping/HectorIterData.msg
[ 75%] Generating Javascript code from hector_mapping/HectorDebugInfo.msg
[ 75%] Built target hector_mapping_generate_messages_nodejs
Scanning dependencies of target hector_mapping_generate_messages
[ 75%] Built target hector_mapping_generate_messages
Scanning dependencies of target hector_mapping
[ 81%] Building CXX object hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/HectorMappingRos.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/build.make:62: recipe for target 'hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/HectorMappingRos.cpp.o' failed
make[2]: *** [hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/HectorMappingRos.cpp.o] Error 4
CMakeFiles/Makefile2:3997: recipe for target 'hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/all' failed
make[1]: *** [hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

If anyone has a clue of how I could use hector_slam on ROS Melodic, I will be very grateful!

jukindle, 2018-09-13:
+1 for this

tiziano3m, 2018-11-07:
+2

4ndr3w, 2018-11-12:
I was able to build hector_slam on ROS Melodic / Ubuntu 18.04 - it's a laptop though, not an embedded platform.

My guess for the original issue post is that `cc1plus` got killed due to OOM on the Beaglebone.

tranz3r, 2019-01-27:
> I was able to build hector_slam on ROS Melodic / Ubuntu 18.04 - it's a laptop though, not an embedded platform.
> 
> My guess for the original issue post is that `cc1plus` was killed due to OOM on the Beaglebone.

Hi, hello, how did you manage to install and execute hector_slam in ROS Melodic on a PC? I use ubuntu 18.04.

atyshka, 2019-03-12:
+1 for this. Hector slam is an important package for the ROS community, Melodic support for this is a must. Binaries would also be appreciated.

StefanFabian, 2019-03-13:
I'm closing this for now since the error message is not due to the code in this repo but most likely an out of memory error on your machine, see [this question on stackoverflow](https://stackoverflow.com/questions/30887143/make-j-8-g-internal-compiler-error-killed-program-cc1plus) for potential solutions.

We are currently evaluating a switch to Melodic and haven't had an issue compiling hector_slam on Melodic.
If one of the other people having issues in this thread has a different compile error, please make a separate issue with the error message.

closed 2019-03-13 by StefanFabian

MaMadDl, 2019-10-26:
i was able to build hector_slam package on odroid XU4 using catkin_tools and with j1 argument