tu-darmstadt-ros-pkg/hector_slam #59: Qt version issue?

atyshka, 2019-03-19:
I'm trying to compile from source on ROS Melodic and the build fails since I don't have QT4. Since Ubuntu 18.04 (and therefore Melodic) uses Qt 5, how can I get this to compile?

GabrielHuettenberger, 2019-03-19:
You can install QT 4 in melodic, using `sudo apt install qt4-default`

closed 2019-05-18 by StefanFabian

rituraj-01, 2020-10-14:
As Ubuntu now uses QT5 by default, you can add QT4 to Ubuntu via third party PPA:
to install QT4-
`sudo add-apt-repository ppa:rock-core/qt4`
`sudo apt update`
`sudo apt install qt4-default`

Now you can catkin_make the hector slam package without the QT error.

nesi73, 2023-02-16:
My environment is Ubuntu 22.04.1 and its worked for me:
`sudo add-apt-repository ppa:ubuntuhandbook1/ppa`