tu-darmstadt-ros-pkg/hector_slam #62: hector_slam package in ros melodic

sandeepmaddula, 2019-05-17:
Hi Can anyone help how to install hector_slam package for Melodic. I cloned hector_slam package and did catkin build, but it is failing to build. Can you suggest how to install hector_slam package for ros-melodic.
Thanks

StefanFabian, 2019-05-18:
I'm sorry but no one can help you solve an issue if you don't provide any information.
I'm assuming you don't have qt4 installed, see #59, but I can't do more than guessing without an error log.

nTnZone, 2019-07-10:
anyone can give a  list of steps to install hector_slam for melodic?

sjamthe, 2019-07-11:
Here are steps that worked for me (on debian linux) after git clone of the source.
1. sudo apt-get install qt4-default
2. catkin_make
The make was successful.

I am using rplidar to generate rostopic /scan. The topic produces message type sensor_msgs/LaserScan
If you look at the header for this topic it has frame_id: "laser"
so I had to modify the following file
hector_mapping/launch/mapping_default.launch
and replace param values for base_frame and odom_frame with "laser"

That is is, then the following launch command showed the map in rviz
roslaunch hector_slam_launch tutorial.launch

ArghyaChatterjee, 2019-08-25:
When I tried to build the package on jetson nano, running on ubuntu 18.04 LTS with ROS melodic desktop version installed, Error shows something like this.

```/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:160:35: error: ‘hector_nav_msgs’ has not been declared
   bool trajectoryProviderCallBack(hector_nav_msgs::GetRobotTrajectory::Request  &req,
                                   ^~~~~~~~~~~~~~~
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:160:81: error: expected ‘,’ or ‘...’ before ‘&’ token
   bool trajectoryProviderCallBack(hector_nav_msgs::GetRobotTrajectory::Request  &req,
                                                                                 ^
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:167:16: error: ‘hector_nav_msgs’ does not name a type
   inline const hector_nav_msgs::GetRobotTrajectoryResponse getTrajectory() const
                ^~~~~~~~~~~~~~~
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:172:37: error: ‘hector_nav_msgs’ has not been declared
   bool recoveryInfoProviderCallBack(hector_nav_msgs::GetRecoveryInfo::Request  &req,
                                     ^~~~~~~~~~~~~~~
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:172:80: error: expected ‘,’ or ‘...’ before ‘&’ token
   bool recoveryInfoProviderCallBack(hector_nav_msgs::GetRecoveryInfo::Request  &req,
                                                                                ^
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:260:3: error: ‘hector_nav_msgs’ does not name a type
   hector_nav_msgs::GetRobotTrajectory::Response trajectory_;
   ^~~~~~~~~~~~~~~
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp: In constructor ‘PathContainer::PathContainer()’:
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:74:118: error: no matching function for call to ‘ros::NodeHandle::advertiseService(const char [11], bool (PathContainer::*)(int), PathContainer*)’
     trajectory_provider_service_ = nh.advertiseService("trajectory", &PathContainer::trajectoryProviderCallBack, this);
                                                                                                                      ^
In file included from /opt/ros/melodic/include/ros/ros.h:45:0,
                 from /home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:30:
/opt/ros/melodic/include/ros/node_handle.h:879:17: note: candidate: template<class T, class MReq, class MRes> ros::ServiceServer ros::NodeHandle::advertiseService(const string&, bool (T::*)(MReq&, MRes&), T*)
   ServiceServer advertiseService(const std::string& service, bool(T::*srv_func)(MReq &, MRes &), T *obj)
                 ^~~~~~~~~~~~~~~~
/opt/ros/melodic/include/ros/node_handle.h:879:17: note:   template argument deduction/substitution failed:
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:74:118: note:   mismatched types ‘MReq&’ and ‘int’
     trajectory_provider_service_ = nh.advertiseService("trajectory", &PathContainer::trajectoryProviderCallBack, this);
                                                                                                                      ^
In file included from /opt/ros/melodic/include/ros/ros.h:45:0,
                 from /home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:30:
/opt/ros/melodic/include/ros/node_handle.h:924:17: note: candidate: template<class T, class MReq, class MRes> ros::ServiceServer ros::NodeHandle::advertiseService(const string&, bool (T::*)(ros::ServiceEvent<MReq, MRes>&), T*)
   ServiceServer advertiseService(const std::string& service, bool(T::*srv_func)(ServiceEvent<MReq, MRes>&), T *obj)
                 ^~~~~~~~~~~~~~~~
/opt/ros/melodic/include/ros/node_handle.h:924:17: note:   template argument deduction/substitution failed:
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:74:118: note:   mismatched types ‘ros::ServiceEvent<MReq, MRes>&’ and ‘int’
     trajectory_provider_service_ = nh.advertiseService("trajectory", &PathContainer::trajectoryProviderCallBack, this);
                                                                                                                      ^
In file included from /opt/ros/melodic/include/ros/ros.h:45:0,
                 from /home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:30:
/opt/ros/melodic/include/ros/node_handle.h:970:17: note: candidate: template<class T, class MReq, class MRes> ros::ServiceServer ros::NodeHandle::advertiseService(const string&, bool (T::*)(MReq&, MRes&), const boost::shared_ptr<X>&)
   ServiceServer advertiseService(const std::string& service, bool(T::*srv_func)(MReq &, MRes &), const boost::shared_ptr<T>& obj)
                 ^~~~~~~~~~~~~~~~
/opt/ros/melodic/include/ros/node_handle.h:970:17: note:   template argument deduction/substitution failed:
/home/arghya/catkin_ws/src/hector_slam/hector_trajectory_server/src/hector_trajectory_server.cpp:74:118: note:   mismatched types ‘MReq&’ and ‘int’
[... cut ...]
hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/build.make:62: recipe for target 'hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/HectorMappingRos.cpp.o' failed
make[2]: *** [hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/HectorMappingRos.cpp.o] Error 1
CMakeFiles/Makefile2:4286: recipe for target 'hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/all' failed
make[1]: *** [hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed```
Can any one help ?

MaMadDl, 2019-10-26:
I suggest  switching to newer catkin workspace ( the one that uses `catkin build` instead of `catkin_make`) 

if you are on embedded try to build with -j1 or -j2 since building hector_slam take too much memory (or add swap file alternatively).

and make sure to have all the dependencies installed (you can find them in `package.xml`)

I've tested this on Odroid XU4 with Ubuntu 18.04 and ROS Melodic

skohlbr, 2019-10-31:
We'll do a release for melodic soon.

henryfhq-drone, 2020-05-13:
Hi, 
I have tried downloading the git hub code to my catkin_ws and I have run the cattkin_make. 

However, I have encoutered an error on Cmake.

This is the picture of the error:

[image]

@skohlbr @StefanFabian I don't know whether did we install wrongly or is there anything else that I need to install?

I'm using a Jetson Nano, Operating System is Ubuntu 18.04 and my ROS is Melodic and I have qt4 installed.

Thank you

StefanFabian, 2020-05-13:
@henryfhq-drone catkin_make is pretty outdated, you should switch to catkin tools if you can.
The actual error you're getting comes from `cv_bridge` not `hector_slam`.
The error message says that it couldn't find the include dirs for opencv, so, check if you have libopencv-dev installed. If it is installed ask the `cv_bridge` maintainers, they might know what could cause this.
Also make sure that all dependencies are installed using `rosdep`.

StefanFabian, 2020-05-13:
Closing this as `hector_slam` is available as a ROS package on melodic for some time now.

closed 2020-05-13 by StefanFabian

sandeepmaddula, 2020-05-13:
@henryfhq-drone  I can understand that you have problems with cv-bridge because of the terminal output. Once try to install (http://wiki.ros.org/cv_bridge) cv_bridge package and check. I have no problems with hector_slam installation. If you have already installed try to remove that package and reinstall package and update all dependencies and check it out.Hopefully it will fix your error

henryfhq-drone, 2020-05-14:
@sandeepmaddula Thank you for your advice. However, as I am still a beginner. Is it possible to show  me how I can download cv_bridge package? I still don't understand how to download the cv_bridge after looking at the wikiros link that you have provided.

StefanFabian, 2020-05-14:
@henryfhq-drone Using ubuntu's package manager you can install it like this:
```
sudo apt install ros-melodic-cv-bridge
```
Alternatively, you can install it from source by cloning the cv_bridge repository into your workspace but you'll have to install the dependencies, e.g., using 'rosdep install cv_bridge`.

However, please ask these kinds of questions on ROS answers. GitHub issues are not the right place for that.

sandeepmaddula, 2020-05-14:
@henryfhq-drone . Clone that cv_bridge package into src folder of your workspace then build your workspace. and install all dependencies using rosdep install package_name

siddharthcb, 2021-01-31:
i am facing the exact same issue as @henryfhq-drone.
 but when i build cv_bridge from source i get a new error related to cv_bridge like this :
```
Base path: /home/slam/catkin_ws
Source space: /home/slam/catkin_ws/src
Build space: /home/slam/catkin_ws/build
Devel space: /home/slam/catkin_ws/devel
Install space: /home/slam/catkin_ws/install
####
#### Running command: "cmake /home/slam/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/slam/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/slam/catkin_ws/install -G Unix Makefiles" in "/home/slam/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/slam/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic

 -- This workspace overlays: /opt/ros/melodic

-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/slam/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built

-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

-- ~~  traversing 19 packages in topological order:
-- ~~  - hector_geotiff_launch
-- ~~  - hector_slam (metapackage)
-- ~~  - hector_slam_launch
-- ~~  - opencv_tests
-- ~~  - hector_map_tools
-- ~~  - hector_nav_msgs
-- ~~  - vision_opencv (metapackage)
-- ~~  - hector_geotiff

-- ~~  - hector_geotiff_plugins
-- ~~  - hector_marker_drawing
-- ~~  - cv_bridge
-- ~~  - image_geometry
-- ~~  - hector_compressed_map_transport
-- ~~  - rplidar_ros
-- ~~  - hector_imu_attitude_to_tf
-- ~~  - hector_imu_tools
-- ~~  - hector_map_server

-- ~~  - hector_trajectory_server
-- ~~  - hector_mapping
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'hector_geotiff_launch'
-- ==> add_subdirectory(hector_slam/hector_geotiff_launch)
-- +++ processing catkin metapackage: 'hector_slam'
-- ==> add_subdirectory(hector_slam/hector_slam)
-- +++ processing catkin package: 

'hector_slam_launch'
-- ==> add_subdirectory(hector_slam/hector_slam_launch)
-- +++ processing catkin package: 'opencv_tests'
-- ==> add_subdirectory(vision_opencv/opencv_tests)
-- +++ processing catkin package: 'hector_map_tools'
-- ==> add_subdirectory(hector_slam/hector_map_tools)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'hector_nav_msgs'
-- ==> add_subdirectory(hector_slam/hector_nav_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- hector_nav_msgs: 0 messages, 5 services

-- +++ processing catkin metapackage: 'vision_opencv'
-- ==> add_subdirectory(vision_opencv/vision_opencv)
-- +++ processing catkin package: 'hector_geotiff'
-- ==> add_subdirectory(hector_slam/hector_geotiff)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'hector_geotiff_plugins'
-- ==> add_subdirectory(hector_slam/hector_geotiff_plugins)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'hector_marker_drawing'

-- ==> add_subdirectory(hector_slam/hector_marker_drawing)
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(vision_opencv/cv_bridge)
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found version "2.7.17") 
CMake Warning at /usr/share/cmake-3.10/Modules/Fin

dBoost.cmake:1626 (message):
  No header defined for python37; skipping header check
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

 Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python37

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)

-- Found OpenCV: /usr (found suitable version "4.1.1", minimum required is "4") found components:  opencv_core opencv_imgproc opencv_imgcodecs 
-- Found ...
```

though i have not installed python3.7 in my system.
what am i doing wrong?

StefanFabian, 2021-02-09:
You may have python 3.7 but it is asking for Boost python.
Make sure you have installed the cv_bridge dependencies using `rosdep install cv_bridge` as @sandeepmaddula suggested.
But I would not recommend building cv_bridge from source. Just install the debian package.
`rosdep install hector_slam` should be all you need if you want to build it from source.
Or go to your workspace src folder and run `rosdep install -i --from-paths .` (the `-i` tells it to ignore package dependencies that are already in your workspace).

I'm locking this now since it attracted lots of unrelated comments.
Please open a new issue if you still have problems but also try to do your own research first since we can not answer all issues that aren't directly related to bugs in our code.