tu-darmstadt-ros-pkg/hector_slam #67: The principle of operation of hector_slam

vectinaut, 2019-09-17:
Hi, I'm trying to understand the principle of operation of hector_slam. Please, give links, where I can see more information about this algorithm. Thanks!

Martin-Oehler, 2019-09-18:
Hi,
please take a look at the paper linked on the [ros wiki page](http://wiki.ros.org/hector_slam).

vectinaut, 2019-09-18:
I mean, how does it (hector_slam) calculate the position, maybe mathematical description, and where do I find this algorithm in this pkg. Thanks!

StefanFabian, 2019-09-18:
For the mathematical description, please take a look at the [paper](https://www.researchgate.net/publication/228852006_A_flexible_and_scalable_SLAM_system_with_full_3D_motion_estimation). Regarding the algorithm, according to the wiki, you can find it in hector_mapping.
I assume it's in the HectorSlamProcessor class.
But be aware that complex algorithms can usually not be pinpointed to a single location.
Please note that neither of us had any part in the development of this software and the original developer is currently otherwise occupied, so you'll, unfortunately, have to find out the rest for yourself.
If you have concrete questions, we can try to help but no guarantees that we will be able to.

closed 2019-09-18 by StefanFabian

vectinaut, 2019-09-18:
I am preparing a project of an Autonomous quadrocopter for admission to the University and I need to describe this positioning system, but in Russia no one develops such positioning systems and I need the help of specialists in this field. I'd be happy if you could help me with that.

StefanFabian, 2019-09-18:
Does it have to be this system?
As far as I know, this approach is limited to 2D and there are better-suited approaches for autonomous vehicles that can move freely in 3D space. E.g. have a look at [this voxblox issue](https://github.com/ethz-asl/voxblox/issues/210) and [mav_voxblox_planning](https://github.com/ethz-asl/mav_voxblox_planning)

vectinaut, 2019-09-18:
So, it is not typical project. I use rplidar, which connected to raspberry pi, and raspberry has connection to flight controller (PixHawk). On raspberry installed mavros and raspberry send position (geometry_msgs/PoseStamped) to pixhawk. If you would leave contact, I send video of flight

StefanFabian, 2019-09-18:
I'm not a fan of having my email exposed to the various spambots crawling the web but you can  contact me using the [contact form](https://stefanfabian.me/contact/) on my website.