tu-darmstadt-ros-pkg/hector_slam #69: I wonder the code of OccGridMapUtil.h

suho0515, 2019-12-02:
`Eigen::Vector3f interpMapValueWithDerivatives(const Eigen::Vector2f& coords)
  {
    //check if coords are within map limits.
    if (concreteGridMap->pointOutOfMapBounds(coords)){
      return Eigen::Vector3f(0.0f, 0.0f, 0.0f);
    }

    //map coords are always positive, floor them by casting to int
    Eigen::Vector2i indMin(coords.cast<int>());

    //get factors for bilinear interpolation
    Eigen::Vector2f factors(coords - indMin.cast<float>());

    int sizeX = concreteGridMap->getSizeX();

    int index = indMin[1] * sizeX + indMin[0];

    // get grid values for the 4 grid points surrounding the current coords. Check cached data first, if not contained
    // filter gridPoint with gaussian and store in cache.
    if (!cacheMethod.containsCachedData(index, intensities[0])) {
      intensities[0] = getUnfilteredGridPoint(index);
      cacheMethod.cacheData(index, intensities[0]);
    }

    ++index;

    if (!cacheMethod.containsCachedData(index, intensities[1])) {
      intensities[1] = getUnfilteredGridPoint(index);
      cacheMethod.cacheData(index, intensities[1]);
    }

    index += sizeX-1;

    if (!cacheMethod.containsCachedData(index, intensities[2])) {
      intensities[2] = getUnfilteredGridPoint(index);
      cacheMethod.cacheData(index, intensities[2]);
    }

    ++index;

    if (!cacheMethod.containsCachedData(index, intensities[3])) {
      intensities[3] = getUnfilteredGridPoint(index);
      cacheMethod.cacheData(index, intensities[3]);
    }

    float dx1 = intensities[0] - intensities[1];
    float dx2 = intensities[2] - intensities[3];

    float dy1 = intensities[0] - intensities[2];
    float dy2 = intensities[1] - intensities[3];

    float xFacInv = (1.0f - factors[0]);
    float yFacInv = (1.0f - factors[1]);

    **return Eigen::Vector3f(
      ((intensities[0] * xFacInv + intensities[1] * factors[0]) * (yFacInv)) +
      ((intensities[2] * xFacInv + intensities[3] * factors[0]) * (factors[1])),
      -((dx1 * xFacInv) + (dx2 * factors[0])),
      -((dy1 * yFacInv) + (dy2 * factors[1]))
    );**
  }
`

Hello sir, I'm studyding hector_mapping code with paper (A Flexible and Scalable SLAM System with Full 3D Motion Estimation).

and i wonder about gradient value of M(Pm), 
the modification of paper is ∇M(Pm) = ( ∂M/∂x(Pm), ∂M/∂y(Pm) ).
but in the code, ∂M/∂y(Pm) is given first and ∂M/∂x(Pm) is given later.
      
∂M/∂y(Pm) = (-((dx1 * xFacInv) + (dx2 * factors[0])))   
∂M/∂x(Pm) = (-((dy1 * yFacInv) + (dy2 * factors[1])))

i guess there is no problem of using hector_mapping with this code.
but i wonder this is correct code or not.

anyway, thanks for the wonderful works of you guys.
i hope i can get any info about it.

regards.

sunmiaozju, 2019-12-09:
I think this is bug

suho0515, 2019-12-11:
> I think this is bug

Thanks for the reply. :)

skohlbr, 2019-12-11:
Note this is a duplicate of #16. I might have a few free cycles over the holidays to go back and look into this.

suho0515, 2019-12-12:
> Note this is a duplicate of #16. I might have a few free cycles over the holidays to go back and look into this.
Thank you for the information! i would close this issue then :)

closed 2019-12-12 by suho0515