tu-darmstadt-ros-pkg/hector_slam #70: hector_compressed_image_transport: Segmentation Fault (core dumped)

CoarseSand, 2020-01-15:
Hello,

I am currently running a custom RL-based exploration algorithm which takes in the map generated by the "gmapping" pkg as an input, in the form of an image. In order to do so, I've used the hector_compressed_image_transport to transform the \map topic generated by gmapping into an image.
However, once a training episode of the exploration algorithm is done, the exploration algorithm moves on to the next episode but the "map_to_image" node shuts down producing the error message: "Segmentation Fault (core dummped)".

I have a hunch this has something to do with the fact that I've tailored the gmapping pkg to reset the map everytime an episode is done, as was done in https://github.com/jukindle/slam_gmapping. On the exploration side, I publish a trigger msg (in this case, 'reset'), for which the gmapping pkg is always listening, upon which the gmapping resets the map. Right after this, there may be a short period of time during which there is no \map node to subscribe to while a new "SlamGmapping" object is being started live again.

StefanFabian, 2020-01-20:
Unfortunately, we currently don't have the resources to investigate this but feel free to make a PR.
If you're not doing that already, you could build the node from source as `Debug` or `RelWithDebInfo` and attach a debugger to find what exactly results in the segmentation fault.
If you can pinpoint the issue to a specific line, we may be able to help.

lianghu3124, 2020-09-14:
there is a bug in the code:
for (int y = size_y_rev; y >= 0; --y){

        int idx_map_y = size_x * (size_y -y);
        int idx_img_y = size_x * y;

        for (int x = 0; x < size_x; ++x){
      ...
it should be:
for (int y = size_y_rev; y >= 0; --y){

        int idx_map_y = size_x * (size_y -y-1);  // here is the change.
        int idx_img_y = size_x * y;

        for (int x = 0; x < size_x; ++x){

StefanFabian, 2020-09-14:
@lianghu3124 Thank you for looking into this!
Could you link the relevant file / line?

meyerj, 2020-09-24:
The issue was already reported in https://github.com/tu-darmstadt-ros-pkg/hector_slam/issues/52.
The bug is in
https://github.com/tu-darmstadt-ros-pkg/hector_slam/blob/a5ac94ba87654664b105520d0641647c95509b6b/hector_compressed_map_transport/src/map_to_image_node.cpp#L116

closed 2021-03-17 by StefanFabian