tu-darmstadt-ros-pkg/hector_slam #74: catkin_make failed undefined reference

Erdemhan, 2020-06-02:
Hi , i cloned hector_slam to my workspace but when i do catkin_make it gives these errors and it fails.
Distro : melodic
tf2 package is already installed
```
[ 97%] Linking CXX executable /home/erdemhan/catkin_ws/devel/lib/tf2_ros/static_transform_publisher
[ 97%] Built target tf2_ros_static_transform_publisher
Scanning dependencies of target hector_mapping
[ 98%] Building CXX object hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/HectorMappingRos.cpp.o
[ 98%] Linking CXX executable /home/erdemhan/catkin_ws/devel/lib/hector_imu_attitude_to_tf/imu_attitude_to_tf_node
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
hector_slam/hector_imu_attitude_to_tf/CMakeFiles/imu_attitude_to_tf_node.dir/build.make:118: recipe for target '/home/erdemhan/catkin_ws/devel/lib/hector_imu_attitude_to_tf/imu_attitude_to_tf_node' failed
make[2]: *** [/home/erdemhan/catkin_ws/devel/lib/hector_imu_attitude_to_tf/imu_attitude_to_tf_node] Error 1
CMakeFiles/Makefile2:11837: recipe for target 'hector_slam/hector_imu_attitude_to_tf/CMakeFiles/imu_attitude_to_tf_node.dir/all' failed
make[1]: *** [hector_slam/hector_imu_attitude_to_tf/CMakeFiles/imu_attitude_to_tf_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 98%] Building CXX object hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/main.cpp.o
[ 98%] Linking CXX executable /home/erdemhan/catkin_ws/devel/lib/hector_imu_tools/pose_and_orientation_to_imu_node
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
hector_slam/hector_imu_tools/CMakeFiles/pose_and_orientation_to_imu_node.dir/build.make:118: recipe for target '/home/erdemhan/catkin_ws/devel/lib/hector_imu_tools/pose_and_orientation_to_imu_node' failed
make[2]: *** [/home/erdemhan/catkin_ws/devel/lib/hector_imu_tools/pose_and_orientation_to_imu_node] Error 1
CMakeFiles/Makefile2:11893: recipe for target 'hector_slam/hector_imu_tools/CMakeFiles/pose_and_orientation_to_imu_node.dir/all' failed
make[1]: *** [hector_slam/hector_imu_tools/CMakeFiles/pose_and_orientation_to_imu_node.dir/all] Error 2
[ 98%] Building CXX object hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/src/PoseInfoContainer.cpp.o
[ 98%] Linking CXX executable /home/erdemhan/catkin_ws/devel/lib/tf2_ros/buffer_server
[ 98%] Linking CXX executable /home/erdemhan/catkin_ws/devel/lib/hector_trajectory_server/hector_trajectory_server
[ 98%] Built target tf2_ros_buffer_server
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
hector_slam/hector_trajectory_server/CMakeFiles/hector_trajectory_server.dir/build.make:118: recipe for target '/home/erdemhan/catkin_ws/devel/lib/hector_trajectory_server/hector_trajectory_server' failed
make[2]: *** [/home/erdemhan/catkin_ws/devel/lib/hector_trajectory_server/hector_trajectory_server] Error 1
CMakeFiles/Makefile2:12120: recipe for target 'hector_slam/hector_trajectory_server/CMakeFiles/hector_trajectory_server.dir/all' failed
make[1]: *** [hector_slam/hector_trajectory_server/CMakeFiles/hector_trajectory_server.dir/all] Error 2
[100%] Linking CXX shared library /home/erdemhan/catkin_ws/devel/lib/libpointcloud_to_laserscan.so
[100%] Built target pointcloud_to_laserscan
[100%] Linking CXX shared library /home/erdemhan/catkin_ws/devel/lib/liblaserscan_to_pointcloud.so
[100%] Built target laserscan_to_pointcloud
[100%] Linking CXX executable /home/erdemhan/catkin_ws/devel/lib/hector_map_server/hector_map_server
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
hector_slam/hector_map_server/CMakeFiles/hector_map_server.dir/build.make:118: recipe for target '/home/erdemhan/catkin_ws/devel/lib/hector_map_server/hector_map_server' failed
make[2]: *** [/home/erdemhan/catkin_ws/devel/lib/hector_map_server/hector_map_server] Error 1
CMakeFiles/Makefile2:12009: recipe for target 'hector_slam/hector_map_server/CMakeFiles/hector_map_server.dir/all' failed
make[1]: *** [hector_slam/hector_map_server/CMakeFiles/hector_map_server.dir/all] Error 2
[100%] Linking CXX executable /home/erdemhan/catkin_ws/devel/lib/hector_mapping/hector_mapping
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/build.make:186: recipe for target '/home/erdemhan/catkin_ws/devel/lib/hector_mapping/hector_mapping' failed
make[2]: *** [/home/erdemhan/catkin_ws/devel/lib/hector_mapping/hector_mapping] Error 1
CMakeFiles/Makefile2:15324: recipe for target 'hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/all' failed
make[1]: *** [hector_slam/hector_mapping/CMakeFiles/hector_mapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

```

StefanFabian, 2020-06-02:
First, I would suggest using catkin tools (catkin build) instead of catkin_make which is deprecated.

Your error suggests `tf2_ros` might not be installed?

Erdemhan, 2020-06-02:
> First, I would suggest using catkin tools (catkin build) instead of catkin_make which is deprecated.
> 
> Your error suggests `tf2_ros` might not be installed?

I tried catkin build too but this time it gives the error when i launch hector_slam (undefined reference).
tf2_ros already installed

Martin-Oehler, 2020-06-03:
Are you building tf from source? Can you try to manually build the tf packages first?

StefanFabian, 2020-06-05:
DId Martins suggestion solve your issue?

Erdemhan, 2020-06-05:
> DId Martins suggestion solve your issue?

I am not able to use my computer now , i will try and let u know asap.

Erdemhan, 2020-06-08:
Thanks , worked.

closed 2020-06-08 by Erdemhan

mn-banjar, 2020-06-30:
I'm facing the same issue. I tried using both catkin tools & catkin_make

/usr/bin/ld: /opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/costmap_2d_node.dir/build.make:157: /home/pi/ros/ros_ws/devel/.private/costmap_2d/lib/costmap_2d/costmap_2d_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:774: CMakeFiles/costmap_2d_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

Erdemhan, 2020-06-30:
> 
> 
> I'm facing the same issue. I tried using both catkin tools & catkin_make
> 
> Errors << costmap_2d:make /home/pi/ros/ros_ws/logs/costmap_2d/build.make.014.log
> /usr/bin/ld: /opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
> collect2: error: ld returned 1 exit status
> make[2]: *** [CMakeFiles/costmap_2d_node.dir/build.make:157: /home/pi/ros/ros_ws/devel/.private/costmap_2d/lib/costmap_2d/costmap_2d_node] Error 1
> make[1]: *** [CMakeFiles/Makefile2:774: CMakeFiles/costmap_2d_node.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....

In may case , I solved it by deleting tf packages and build files from my workspace and reinstalled tf/tf2 packages with apt-get

mn-banjar, 2020-07-02:
> > I'm facing the same issue. I tried using both catkin tools & catkin_make
> > Errors << costmap_2d:make /home/pi/ros/ros_ws/logs/costmap_2d/build.make.014.log
> > /usr/bin/ld: /opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
> > collect2: error: ld returned 1 exit status
> > make[2]: *** [CMakeFiles/costmap_2d_node.dir/build.make:157: /home/pi/ros/ros_ws/devel/.private/costmap_2d/lib/costmap_2d/costmap_2d_node] Error 1
> > make[1]: *** [CMakeFiles/Makefile2:774: CMakeFiles/costmap_2d_node.dir/all] Error 2
> > make[1]: *** Waiting for unfinished jobs....
> 
> In my case , I solved it by deleting tf packages and build files from my workspace and reinstalled tf/tf2 packages with apt-get
> EDIT: if your tf packages in your workspace try to delete from workspace and install to your ros source with apt-get install or apt install

Thank you, it works
I deleted all the packages and reinstalled it again without repetition in my workspace, I have already some tf packages in /opt/ros/melodic/share