tu-darmstadt-ros-pkg/hector_slam #76: Noetic release

mintar, 2020-06-17:
Could you release hector_slam into ROS Noetic, please?

StefanFabian, 2020-06-17:
We probably will once we've started making the switch which will not happen before moveit and other major packages are officially released.

mintar, 2020-06-17:
Thanks for the quick response! This is not super urgent for me, I can be patient. :)

Timple, 2020-11-23:
I guess most major packages are officially released now :slightly_smiling_face:

StefanFabian, 2020-11-23:
True, however, with the switch to noetic, I've also taken the time to port `hector_geotiff` from Qt4 to Qt5 which was quite a bit harder than it sounds.
Could use a bit more testing and I also vaguely remember there being a cyclic dependency issue I wanted to fix before releasing on noetic.

StefanFabian, 2020-12-02:
@mintar @Timple and anyone who is interested.
I have pushed a clean up commit on noetic-devel.
If you guys could confirm that whatever you're using hector_slam for still works, I will prepare a noetic release.

mintar, 2020-12-02:
I've tested my use case ([hector_mapping on the MiR robot](https://github.com/dfki-ric/mir_robot#gazebo-demo-mapping)) and can confirm that it works on noetic. I only use the `hector_mapping` part, so I haven't tested any of the other stuff, like `hector_geotiff`.

StefanFabian, 2020-12-17:
Alright, I've just made a release. Could take a while until it's synced, though.

closed 2020-12-17 by StefanFabian