tu-darmstadt-ros-pkg/hector_slam #78: data structure of topic /map

lukelu0520, 2020-07-09:
Dear tu-darmstadt-ros-pkg,

Thanks for your hector_slam work, I've successfully ran and visualized /map, /trajetory, /slam_out_pose via RViz. It looks very nice.

Now I'd like to 'see' numerical data of these topics and check if it makes numerical sense, so I did rostopic echo /map, and print it out to a .bag and .csv file.

However, it shows all "-1" in field.data# (# is fom 0 to 1008).  I was wondering how to interpret /map topic data.
Maybe I miss something, hope to hear your expertise on the data structure and definition.

Thank you.

Martin-Oehler, 2020-07-09:
The message documentation can be found [here](http://docs.ros.org/melodic/api/nav_msgs/html/msg/OccupancyGrid.html).

lukelu0520, 2020-07-09:
Problem solved~!

The map size is default to set 2048, but my occupancy area is too small (~1%) when compared to the map area. So it is almost invisible by eyes. (but I don't know why echo > csv doesn't print entire array out)

After changing the map size to a smaller value, I could see data within [0, 100] value.

closed 2020-07-09 by lukelu0520