tu-darmstadt-ros-pkg/hector_slam #81: Port hector_geotiff to Qt5 on headless systems

StefanFabian, 2020-09-12:
The current noetic-devel branch will not work on robot platforms as they usually do not run an X server which is required for the Qt5 components used.
One way to get around this seems to be platforms but further investigation is needed.

Tristan9497, 2020-10-29:
i have the same issue while performing cmake in ros noetic fails at hector_geotiff with the following msg
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------

CMake Error at /usr/share/cmake-3.16/Modules/FindQt4.cmake:1314 (message):
  Found unsuitable Qt version "5.12.8" from /usr/bin/qmake, this code
  requires Qt 4.x

StefanFabian, 2020-12-02:
@Tristan9497 That should be fixed now. The noetic-devel works with Qt5 and was tested on a headless system.

closed 2020-12-02 by StefanFabian

Tristan9497, 2020-12-03:
> @Tristan9497 That should be fixed now. The noetic-devel works with Qt5 and was tested on a headless system.

Thanks for the Update i'll give it a try in a couple of days