tu-darmstadt-ros-pkg/hector_slam #85: Error has occurred while running 'roslaunch hector_slam_launch tutorial.launch'

darkasassin44, 2021-01-13:
Maybe more than a month earlier(It was 13 Dec 2020 noted on my scheduler), I've done this process successfully without any error.
I've followed this Youtube video : [https://www.youtube.com/watch?v=Qrtz0a7HaQ4](url)

1st terminal : `roscore`
2nd terminal : `roslaunch rplidar_ros rplidar_a3.launch`
3rd terminal : `roslaunch hector_slam_launch tutorial.launch`
I could find an error on 3rd terminal.
```
lbd_song@lbd-gcs:~$ roslaunch hector_slam_launch tutorial.launch 
... logging to /home/lbd_song/.ros/log/bcdf61de-5540-11eb-a4ce-e0d4e8ff40e7/roslaunch-lbd-gcs-12887.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: while processing /home/lbd_song/catkin_ws/src/hector_slam/hector_geotiff/launch/geotiff_mapper.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/lbd_song/catkin_ws/src/hector_slam/hector_geotiff/launch/geotiff_mapper.launch'
The traceback for the exception was written to the log file
```
I cannot find the directory as error says too.
Please tell me how to fix this error.
Thank you all the time.

StefanFabian, 2021-01-13:
Thank you for the report.
The geotiff launch files were moved to a separate package, looks like I've missed a reference.

matthias-schleibinger, 2021-01-14:
If I am not mistaken the "geotiff_mapper.launch" was moved from the "hector_geotiff" package to the "hector_geotiff_launch" package. So in order to resolve this error you have to go to your "tutorial.launch" file and change the line
`<include file="$(find hector_geotiff)/launch/geotiff_mapper.launch">`
to
`<include file="$(find hector_geotiff_launch)/launch/geotiff_mapper.launch">`

StefanFabian, 2021-01-15:
Should be fixed by 77b51a6.

closed 2021-01-15 by StefanFabian

Iulian19768, 2022-12-05:
Hello, i am getting this error when running : "roslaunch hector_slam_launch tutorial.launch" 
`Resource not found: The following package was not found in <arg default="$(find hector_geotiff)/maps" name="geotiff_map_file_path"/>: hector_geotiff
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/robotica/sdk_ld06_raspberry_ros/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file
`