tu-darmstadt-ros-pkg/hector_slam #90: Cache static tf transforms from base to lidar frame(s) at node startup

marcelino-pensa, 2021-03-22:
I have an application that is very sensitive to tf tree lookups. Our ROS master does not run on our robot and our network can be unreliable or slow to respond.
Because of that, it is really impractical for us to have tf lookups at every `scanCallback`.
Hence, I would request to have a process to cache static tf transforms at node startup so that the tf tree does not need to be checked all the time.

Martin-Oehler, 2021-03-23:
It should not be a concern, where the ROS master is running as long as the tf node is on the robot as well.

StefanFabian, 2021-03-23:
I agree with Martin, you should be very certain this is actually (the source of) a problem for you.
If I'm not mistaken, the communication with the ROS master is only to establish a connection on subscription, and afterward, they should have a direct connection.
Tf also works using subscribers, so, the lookup itself will not lead to any network requests.
I've had a look at tf2's source code and it appears that if the transform you are looking up is made of static transforms `waitForTransform` should even return immediately without waiting for the transforms at the given timestamp.

closed 2021-03-23 by StefanFabian

marcelino-pensa, 2021-03-23:
Thanks for your inputs, @Martin-Oehler and @StefanFabian!
I will do further investigation, but we definitely had some trouble with previous third-party code that we used. According with our logs, we would have the error:
```
[ERROR] [1586521959.047313662]: [registerPublisher] Failed to contact master at [basestation:11311].  Retrying...
```

I can think of alternative root-problems, such as:
- Distinctions between lookup in `tf` vs `tf2`
- ROS version
- Redeclaration of the listener (each declaration of that would need to communicate with master

I will do some investigation to determine whether the issue actually occurs with Hector running in Noetic. If so, I might reopen the issue