tu-darmstadt-ros-pkg/hector_slam #92: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'

quangvu16, 2021-03-27:
Dear Friends,
My system: Jetson Nano, Ros Melodic Desktop Full.
When I catkin_make this package, it shows below issue. Please help me. Thank you.

[ 94%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/hector_imu_attitude_to_tf/imu_attitude_to_tf_node
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
RPLidar_Hector_SLAM/hector_slam/hector_imu_attitude_to_tf/CMakeFiles/imu_attitude_to_tf_node.dir/build.make:118: recipe for target '/home/ubuntu/catkin_ws/devel/lib/hector_imu_attitude_to_tf/imu_attitude_to_tf_node' failed
make[2]: *** [/home/ubuntu/catkin_ws/devel/lib/hector_imu_attitude_to_tf/imu_attitude_to_tf_node] Error 1
CMakeFiles/Makefile2:6308: recipe for target 'RPLidar_Hector_SLAM/hector_slam/hector_imu_attitude_to_tf/CMakeFiles/imu_attitude_to_tf_node.dir/all' failed
make[1]: *** [RPLidar_Hector_SLAM/hector_slam/hector_imu_attitude_to_tf/CMakeFiles/imu_attitude_to_tf_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 95%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/hector_imu_tools/pose_and_orientation_to_imu_node
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
RPLidar_Hector_SLAM/hector_slam/hector_imu_tools/CMakeFiles/pose_and_orientation_to_imu_node.dir/build.make:118: recipe for target '/home/ubuntu/catkin_ws/devel/lib/hector_imu_tools/pose_and_orientation_to_imu_node' failed
make[2]: *** [/home/ubuntu/catkin_ws/devel/lib/hector_imu_tools/pose_and_orientation_to_imu_node] Error 1
CMakeFiles/Makefile2:6364: recipe for target 'RPLidar_Hector_SLAM/hector_slam/hector_imu_tools/CMakeFiles/pose_and_orientation_to_imu_node.dir/all' failed
make[1]: *** [RPLidar_Hector_SLAM/hector_slam/hector_imu_tools/CMakeFiles/pose_and_orientation_to_imu_node.dir/all] Error 2
[ 95%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/tf2_ros/buffer_server
[ 95%] Built target tf2_ros_buffer_server
[ 95%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/hector_map_server/hector_map_server
/opt/ros/melodic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
RPLidar_Hector_SLAM/hector_slam/hector_map_server/CMakeFiles/hector_map_server.dir/build.make:118: recipe for target '/home/ubuntu/catkin_ws/devel/lib/hector_map_server/hector_map_server' failed
make[2]: *** [/home/ubuntu/catkin_ws/devel/lib/hector_map_server/hector_map_server] Error 1
CMakeFiles/Makefile2:6480: recipe for target 'RPLidar_Hector_SLAM/hector_slam/hector_map_server/CMakeFiles/hector_map_server.dir/all' failed
make[1]: *** [RPLidar_Hector_SLAM/hector_slam/hector_map_server/CMakeFiles/hector_map_server.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

marcelino-pensa, 2021-04-06:
Did you try `sudo apt install ros-melodic-tf2-ros`?

quangvu16, 2021-04-06:
I solved this. Thank you

closed 2021-04-07 by StefanFabian

ju-mingyue, 2021-04-25:
@quangvu16 I am sorry that I also encountered this problem. I would like to ask you how to solve it?

quangvu16, 2021-04-25:
1) If you install from this github, you should choose devel (noetic, kniectic, kinetic) which you install 

2) The other simple way is to isntall as below command.
Noetic:
sudo apt install ros-noetic-tf2-ros
Melodic:
sudo apt install ros-melodic-tf2-ros
Kinetic:
sudo apt install ros-kinetic-tf2-ros
After that you use sudo nano to change parameters.
https://ardupilot.org/dev/docs/ros-slam.html

bears0, 2022-01-22:
I solved this by making sure that I cloned the melodic-devel branch. I'm running on an RPI4 so I have to compile everything from source. I've found that its best to check github for available branches before blindly cloning.

davidschep, 2022-03-14:
Hi unfortunately after looking at this issue and similar posts I still run into this problem.

I use `catkin build`, building the `melodic-devel` branch (on an ubuntu 18.04 machine), I tried `noetic-devel` too just to see. `ros-melodic-tf2-ros` is installed (and ros completely reinstalled using apt-get to see if that would solve it).

Here is the full console output: [https://pastebin.com/iqD4GMAg](https://pastebin.com/iqD4GMAg)

What other solutions could I try?

marcelino-pensa, 2022-03-14:
David, it doesn't seem like your error is anything like the OP error
Looking at your console output it seems like there is something wrong with your ROS installation. For instance this error:
```
/opt/ros/melodic/include/tf/tf.h:403:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   std::shared_ptr<tf2_ros::Buffer> tf2_buffer_ptr_;
```

It seems to me that ROS isn't properly installed in your device, as it can't compile the native library `tf.h`. Did you modify any of `tf.h`? I don't know how wouldn't it find `shared_ptr` as part of `std`

I don't have much of an idea of how you got there, but a few things to try might be to reinstall `tf` (`apt install ros-melodic-tf`) or just trying to purge all of ROS and reinstalling, since other libraries might also be compromised

StefanFabian, 2022-03-14:
Shared ptr was introduced in C++11 ([Docs](https://en.cppreference.com/w/cpp/memory/shared_ptr)).
Hence, it makes sense that it won't compile with older C++ standards.
Since Melodic allows up to C++14, I would be okay with adding `set(CMAKE_CXX_STANDARD 11)` which is preferable over `add_compile_options` if it is available (requires CMake version >=3.1, but I think Ubuntu 18.04 comes with 3.5.2) and updating the min required cmake version accordingly.