tu-darmstadt-ros-pkg/hector_slam #94: Hector Slam metapackage not available for Noetic through apt?

cailloumax, 2021-04-08:
More of a question than an issue: 
I work with both Melodic and Noetic, using respectivly a Jetson Nano (Ubuntu 18.04 aarch64) and a VM (Ubuntu 20.04 amd64). I develop a metapackage on my own that depend on the _hector-slam_ and that will launch the various sensor and the SLAM stack. It works fine with the Melodic but when I use it on the Noetic, I was not able to install  _hector-slam_ through apt.

I tried to compile it myself, but then, while compiling is fine, the dependencies of my own package are not fulfilled as the cloned repository does not provide the CMake of the metapackage.

I may explicitly require every package in the stack, instead of the metapackage, but is there an easier solution ?

StefanFabian, 2021-04-08:
Ah, thank you for bringing this to our attention. It looks like the build farm has trouble building the debian package for hector_geotiff_plugins, hence, all dependants are not available.
I will look into this.

StefanFabian, 2021-04-16:
I have done another release and it looks like it is building now.
Will have to wait for the next sync before it is available, though.

closed 2021-04-16 by StefanFabian