tu-darmstadt-ros-pkg/hector_slam #95: Saving Geotiff map issue

kompaso, 2021-04-20:
Hello!

I'm trying to save a map using tutorial provided here 

http://wiki.ros.org/hector_slam/Tutorials/MappingUsingLoggedData

After I run command: rostopic pub syscommand std_msgs/String "savegeotiff"
 I receive following:
publishing and latching message. Press ctrl-C to terminate

At the same time, in the directory <.../hector_slam/hector_geotiff/maps > appears the only file with extension TFW. How can  find a TIF file? What is the problem?

Ubuntu 20.04 + ROS NOETIC.

I have tried another setup: Ubuntu 18.04 + ROS MELODIC. But in such a way it generates nothing (nor TIF, nor TFW).

Hope for your help.

Thank you!

poly-tropos, 2021-12-16:
same issue here... any suggestion?

StefanFabian, 2021-12-16:
Can you check that you have the following debian package installed: `qt5-image-formats-plugins`?

poly-tropos, 2021-12-20:
> Can you check that you have the following debian package installed: `qt5-image-formats-plugins`?

Indeed, that package was missing, now it works, thanks!