tu-darmstadt-ros-pkg/hector_slam #101: Cannot locate node of type [hector_mapping]

User852369, 2022-01-22:
I have installed Ros Noetic under Ubuntu 20.04.3 LTS. With the Slamtec RPLIDAR A1 I can start the Rviz without any problems. After cloning the reposetory in "/home/ubuntu/catkin_ws/src" and running "roslaunch rplidar_ros rplidar.launch" in terminal 1 and "roslaunch hector_slam_launch tutorial.launch" in terminal 2 the following errors occur:

'''
process[rviz-1]: started with pid [5194]
ERROR: cannot launch node of type [hector_mapping/hector_mapping]: Cannot locate node of type [hector_mapping] in package [hector_mapping]. Make sure file exists in package path and + is set to executable (chmod +x)
process[base_to_laser_broadcaster-3]: started with pid [5195]
ERROR: cannot launch node of type [hector_trajectory_server/hector_trajectory_server]: Cannot locate node of type [hector_trajectory_server] in package [hector_trajectory_server]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [hector_geotiff/geotiff_node]: Cannot locate node of type [geotiff_node] in package [hector_geotiff]. Make sure file exists in package path and permission is set to executable (chmod +x)
'''

But with the roscd command I can reach all areas.

The Rviz gui starts but does not create a map (also no measuring points are displayed). The application also shows a Global Status Error.

What can I do to start the tutorial , that a map is created?

RajithAshok, 2024-03-21:
Hi @PhlpGr , I'm using ROS Melodic on Ubuntu 18.04 and facing the same error while running the command "roslaunch hector_slam_launch tutorial.launch". If you have already solved the error, your insights on the same will be very helpful.
Thank you

Maleby, 2024-09-09:
Hello, I am using ROS Noetic and I am having the same issue. I would appreciate if someone has an idea on how to solve this!

RajithAshok, 2024-09-09:
Hi @Maleby , I solved the issue with the below command 

`catkin_make --only-pkg-with-deps <target_package>`

Make sure all the files inside the packages are executable and run this command for all the packages individually. 
For more info visit: [how-to-build-just-one-package-using-catkin_make](https://answers.ros.org/question/54178/how-to-build-just-one-package-using-catkin_make/)

Hope this helps.